GyverLibs/GyverOLED #9: Скалирование  неприятный эффект SSD1306

schpuntik, 2022-01-24:
Если  скалировать   цифры  то  появляется  неприятный эффект.  В моём   случае сделал тахометр    всё работало нормально. Решил увеличить размер вывода   и получается  что первый  столбец   пикселей  пропадает   или мигает . Чем  больше скалирование тем больше столбцов пропадает.  Без  скалирования выводится нормально.  Дисплей SSD1306  128x64  no buffer.

GyverLibs, 2022-01-24:
Можно минимальный пример, в котором происходит баг? Используется ли буфер? В какие координаты производится вывод?

schpuntik, 2022-01-24:
https://youtu.be/fGf1oUptcVM  Здесь можно посмотреть

linotex, 2022-01-24:
Алекс имел ввиду пример кода чтобы у себя попробовать.

schpuntik, 2022-01-24:
Без проблем могу и код выложить.

GyverLibs, 2022-01-24:
> Без проблем могу и код выложить. Полностью или только кусок который выводит на дисплей?

Я вроде четко написал что требуется)

schpuntik, 2022-01-24:
Я в первом посте написал  какой дисплей   и что  буффер не используется.  Вот кусок кода   который  выводится тахометр. Если  этого  мало то могу  и весь код  выложить.
`void datChek() {
  per =  millis() - timer0;
  timer0 = millis();

}

int realSpeed() {
  oled.setScale(3);
  if ((millis() - timer0) < 1000) {
    int nowSpeed;
    nowSpeed = 60000 / per;
    oled.setCursorXY(49, 38);
    delay(100);
    oled.print(nowSpeed);
    oled.print("     ");
    return (nowSpeed);
  } else {
    oled.setCursorXY(49, 38);
    oled.print("СТОП");
    return (0);
  }

}`

linotex, 2022-01-24:
В идеальном случае нужно выкладывать код который можна Ctrl+C && Ctrl+V и он будеть работать. К примеру у меня нету переменной `per`, соответственно ничего не компилируется(((

linotex, 2022-01-24:
Также если ви просто скопируете полностю ваш код - ничего не вийдет потому что у вас еще много обвязки (возможно Алекс захочеть собирать это но врятли). Но я думаю данные можна симулировать

linotex, 2022-01-24:
Минимальний код
```
#include <GyverOLED.h>

static GyverOLED<SSD1306_128x64, OLED_NO_BUFFER> oled;
unsigned long timer0 = 0;
void setup()
{
  oled.init();
      oled.setContrast(255);
      oled.clear();
}

void loop() { 
  if ((millis() - timer0) >= 1000) {
    oled.setScale(3);
    oled.setCursorXY(49, 38);
    timer0 = millis();

    oled.print(timer0);
    oled.print(" ");
  }
}```
`

linotex, 2022-01-24:
У меня проблема возникает когда `timer0 >= 20000`

linotex, 2022-01-24:
PS: Єбать Остапа понесло

schpuntik, 2022-01-24:
https://drive.google.com/file/d/1ixZECcnbqOVe-I4fnnsm5Y7u781JMYKf/view?usp=sharing
Здесь полностью  код.    Я  понял  так что   ему важно  знать  в каком месте выводится это на дисплей.   Я  не  программист  просто  учусь  и не знаю  всех нюансов.  Я просто   взял прошивку   другого  человека и  правлю под мои хотелки.

schpuntik, 2022-01-24:
Факт  то  что   без маштабирования всё  нормльно   показывает на дисплее. Но для меня  это  слишком мелкий шрифт.  Как начинаю  увиличивать  шрифт  сразу  появляются  артефакты.   Алекс  говорил мне  что без буфера  надо  всегда  кратным 8  пикселям  делать. Так вот я пробывал  двигать  вывод  по дисплею   кратным 8 пикселям  но ничего  не получилось.

linotex, 2022-01-24:
Но у  вас `setCursor` не кратный 8. Установите `48, 32` к примеру. У меня пофиксилось

linotex, 2022-01-24:
Заработало или нет? Я только потом увидел что ви уже пробовали устанавливать кратно 8, но надеялся что пофикситься :D

schpuntik, 2022-01-24:
По Х 48  я пробывал  а вот по У   не помню, вроде тоже. А вот одновременно  ХУ  наверно нет.  У спустил на 40  Так вроде лутьше смотрится. Спасибо!

linotex, 2022-01-24:
Кул

linotex, 2022-01-24:
Если проблема решена - пожалуста закройте ишю). Спасибо

closed 2022-01-24 by schpuntik