GyverLibs/GyverOLED #12: Глюки при отображении картинок на экране sh1106 128x64

Denis20221, 2022-04-05:
Это происходит не со всеми картинками, но в основном с теми что по крупней(бывало и с мелкими иконками). Под картинкой появляется белая полоса, или весь экран заполняется хаотично символами и т.д. Запускал из стандартного примере oled_demo, sh1106 128x64px i2c. Картинку конвертировал через вложенную программу к вашей  библиотеки.

Вот пример картинки с которой вылазиет глюк:
https://disk.yandex.ru/i/mQPA-3yG8c5m3A

А вот так это выглядит:
https://disk.yandex.ru/i/C-gArP0gmUdC9w

Denis20221, 2022-04-06:
P.S. Забыл добавить что i2c использовался стандартный, скорость 400кгц.
Мозги raspberry pi pico, ядро-ардуино earlephilhower [https://github.com/earlephilhower/arduino-pico](url)

GyverLibs, 2022-04-06:
Привет, возможно какие то проблемы с пико, уже не раз сталкивался с неадекватной работой кода, который отлично работает на всех ардуинах

ddmm2005, 2023-03-01:
У меня также была полоса только на Atmega328p. Воспользовался онлайн конвертером на сайте Duino. Картинка после этого стала нормально отображаться. На сайте правда конвертер выдает инвертированный код. Поэтому сначала инвертировал картинку и потом туда загрузил.