GyverLibs/GyverOLED #14: Arduino IDE 2.0.0 не компилирует для esp32

Allexnero, 2022-10-02:
После выхода 2 версии IDE начал потихоньку переносить свои проекте. И проект с GyverOLED отказался компилироваться. Начал разбираться и понял что примеры из библиотеки тоже не копулируются. При чем на Arduino IDE 1.8.19 все работает нормально. Везде стоит GyverOLED 1,5 и менеджер платы esp32 1.0.6 отличия только в IDE. Пробовал компилировать ардуино и esp8266 все получается, а вот esp32 не хочет.  Вначале подумал на кириллицу в адресе, но другие библиотеки работают нормально , да и проблема только с  компиляцией для esp32. Мне кажется что в новой IDE по другому компилирует и то что раньше "некоторые предупреждения" допускались, то с новой "интерпретируются как ошибки" и не дают компилироваться.

Версия: IDE2.0.0
Дата: 2022-09-14T07:06:37.759Z
Версия CLI: 0.27.1 [a900cfb2]

лог

In file included from C:\Users\Администратор\AppData\Local\Temp\.arduinoIDE-unsaved202292-2916-rtwq2w.nrhhp\oled_demo\oled_demo.ino:7:0:
c:\Users\�������������\Documents\Arduino\libraries\GyverOLED\src/GyverOLED.h: In instantiation of 'void GyverOLED<_TYPE, _BUFF, _CONN, _CS, _DC, _RST>::fastLineH(int, int, int, byte) [with int _TYPE = 2; int _BUFF = 1; int _CONN = 0; signed char _CS = -1; signed char _DC = -1; signed char _RST = -1; byte = unsigned char]':
C:\Users\Администратор\AppData\Local\Temp\.arduinoIDE-unsaved202292-2916-rtwq2w.nrhhp\oled_demo\oled_demo.ino:114:26:   required from here
c:\Users\�������������\Documents\Arduino\libraries\GyverOLED\src/GyverOLED.h:412:28: error: operation on 'x1' may be undefined [-Werror=sequence-point]
         if (x0 > x1) _swap(x0, x1);
                            ^
c:\Users\�������������\Documents\Arduino\libraries\GyverOLED\src/GyverOLED.h:147:21: note: in definition of macro '_swap'
 #define _swap(x, y) x = x ^ y ^ (y = x)
                     ^
c:\Users\�������������\Documents\Arduino\libraries\GyverOLED\src/GyverOLED.h: In instantiation of 'void GyverOLED<_TYPE, _BUFF, _CONN, _CS, _DC, _RST>::fastLineV(int, int, int, byte) [with int _TYPE = 2; int _BUFF = 1; int _CONN = 0; signed char _CS = -1; signed char _DC = -1; signed char _RST = -1; byte = unsigned char]':
C:\Users\Администратор\AppData\Local\Temp\.arduinoIDE-unsaved202292-2916-rtwq2w.nrhhp\oled_demo\oled_demo.ino:116:26:   required from here
c:\Users\�������������\Documents\Arduino\libraries\GyverOLED\src/GyverOLED.h:441:28: error: operation on 'y1' may be undefined [-Werror=sequence-point]
         if (y0 > y1) _swap(y0, y1);
                            ^
c:\Users\�������������\Documents\Arduino\libraries\GyverOLED\src/GyverOLED.h:147:21: note: in definition of macro '_swap'
 #define _swap(x, y) x = x ^ y ^ (y = x)
                     ^
c:\Users\�������������\Documents\Arduino\libraries\GyverOLED\src/GyverOLED.h: In instantiation of 'void GyverOLED<_TYPE, _BUFF, _CONN, _CS, _DC, _RST>::rect(int, int, int, int, byte) [with int _TYPE = 2; int _BUFF = 1; int _CONN = 0; signed char _CS = -1; signed char _DC = -1; signed char _RST = -1; byte = unsigned char]':
C:\Users\Администратор\AppData\Local\Temp\.arduinoIDE-unsaved202292-2916-rtwq2w.nrhhp\oled_demo\oled_demo.ino:117:27:   required from here
c:\Users\�������������\Documents\Arduino\libraries\GyverOLED\src/GyverOLED.h:483:28: error: operation on 'x1' may be undefined [-Werror=sequence-point]
         if (x0 > x1) _swap(x0, x1);
                            ^
c:\Users\�������������\Documents\Arduino\libraries\GyverOLED\src/GyverOLED.h:147:21: note: in definition of macro '_swap'
 #define _swap(x, y) x = x ^ y ^ (y = x)
                     ^
c:\Users\�������������\Documents\Arduino\libraries\GyverOLED\src/GyverOLED.h:484:28: error: operation on 'y1' may be undefined [-Werror=sequence-point]
         if (y0 > y1) _swap(y0, y1);
                            ^
c:\Users\�������������\Documents\Arduino\libraries\GyverOLED\src/GyverOLED.h:147:21: note: in definition of macro '_swap'
 #define _swap(x, y) x = x ^ y ^ (y = x)
                     ^
cc1plus.exe: some warnings being treated as errors

exit status 1

Compilation error: exit status 1

P.S. Во время написания Issues решил погонять другие библиотеки гайвера которые у меня стояли и похожая проблема c esp32 1.0.6 возникла в новой среде с GyverTM1637 хотя лог немного другой. Возможно проблемы есть  с другими старыми библиотеками. 
GyverMotor я так понял вообще не поддерживает esp32 
А вот с EncButton, GyverStepper, GyverPortal проблем никаких не было.

Проблема не существенная и легко обходиться, но для новичков которые начинают с 2.0 могут  не понять что проблема в IDE

GyverLibs, 2022-10-02:
Все библиотеки поддерживают esp32, кроме тех где написано "только avr". Это directTimers, directADC, GyverTimers, GyverWDT и возможно что то ещё. Всё остальное должно работать, если не работает - это специфика индийского компилятора и нужно сидеть фиксить

GyverLibs, 2022-12-19:
компилится под esp32 на IDE v2, возможно после обновления 1.6

closed 2022-12-19 by GyverLibs