GyverLibs/GyverOLED #15: Шум на экране SSD1306 при подключении через SPI вместе с адаптером для SD карт

fagirton, 2022-11-05:
SPI экран модели SSD1306 при одновременном подключении с адаптером microSD карт выдает случайные пиксели по всему экрану (похоже на белый шум). Проверил оба элемента отдельно - работают исправно, но вместе нет

Может ли это быть из-за особенностей поддержки SPI в библиотеке?

GyverLibs, 2022-11-05:
Схему подключения как говорится в студию)

fagirton, 2022-11-05:
Arduino Nano:
SCK (Дисплей и Картридер) < D13   D12 > MISO (Картридер)
                                                                   D11 > SDA (Дисплей) и MOSI (Картридер)

                                                                   D8 > CS (Дисплей)
                                                                   D7 > DC (Дисплей)
                                                                   D6 > RES (Дисплей)

                                                                   D4 > CS (Картридер)

                 Внешнее питание + < 5V
                 Внешнее питание - < GND

Картридер и дисплей подключены к 3.3V, на том же внешнем питании

GyverLibs, 2022-11-05:
Очень странно. cs подключены к разным пинам, общение с картой памяти не должно перекрещиваться с дисплеем. Чисто на всякий случай - а если отключить cs дисплея от ардуины?

fagirton, 2022-11-05:
Проверил, результат тот же

GyverLibs, 2022-11-05:
Вот это уже похоже на эзотерику или кривую разводку платы дисплея. По сути с неприжатым cs дисплей вообще не должен реагировать на команды на шине

GyverLibs, 2022-11-29:
В общем это известная беда SD карт http://elm-chan.org/docs/mmc/mmc_e.html, пункт Cosideration on Multi-slave Configuration. Как решить: после завершения общения с картой памяти нужно отпустить CS карты (библа SD возможно сама это делает, либо отпустить вручную) и закинуть по SPI пару байт (пару нулей условно). Почему - карта держит линию даты

closed 2022-11-29 by GyverLibs