GyverLibs/GyverOLED #16: Добавить логирование

osintaras, 2022-11-10:
Добрый день! Все прекрасно, но хорошо бы чтобы был функционал вывода на экран лога - чтобы по мере добавления новых строк, старое затиралось и картинка двигалась вверх. В общем чтобы можно было выводить логи работы устройства.

GyverLibs, 2022-11-10:
Добрый, это программисту лучше сделать самостоятельно, потому что в рамках библиотеки придётся хранить целый буфер с текстом. Что утяжелит библиотеку и в этом нет смысла

closed 2022-11-10 by GyverLibs

osintaras, 2022-11-14:
А есть методы чтобы понять текущее положение курсора на экране? Причем чтобы на экране еще ничего не появилось?
Типа byte getCursorX() и byte getCursorY(). Иначе как понять, что строка закончилась? 
В строке ведь не фиксированное количество символов - шрифт не моноширинный. А если и моноширинный, то все бы хорошо, но для кириллицы нет substring в ардуино...

GyverLibs, 2022-11-14:
Есть то, что описано в документации. Гет курсор могу добавить, это не проблема. Также в библиотеке есть автопринтлн, который позволяет автоматически переносить строку если она закончилась, а также сигнал о том что курсор упёрся в нижний правый угол (конец дисплея)