GyverLibs/GyverOLED #19: Поддержка прокрутки текста

n0t-amused, 2022-12-06:
Было бы классно иметь библиотечный способ для горизонтальной прокрутки текста (бегущей строки).

ddmm2005, 2023-03-01:
Поддерживаю. Приходится костылями делать через for

GyverLibs, 2023-03-01:
Хосподе, это ведь в три строчки делается в основной программе. Библиотеку это утяжелит, придется добавлять настройки бега строки, чтобы всем угодить

GyverLibs, 2023-03-01:
Плюс не очень понятно, откуда будет браться текст. То есть надо держать у себя в программе буфер с текстом и указатель на него передавать в библиотеку

ddmm2005, 2023-03-01:
Да не спорю, вес больше. Но можно реализовать вкл/выкл через define. Как со шрифтами.

ddmm2005, 2023-03-01:
Библиотека отличная, перешел на нее с OLED_I2C. Спасибо!

GyverLibs, 2023-03-01:
Подумаю, как можно удобно сделать

Ivannno, 2025-01-29:
А почем в oledMenu при прокрутке через строчку Exit сбрасываются значения, которые редактировал?