GyverLibs/GyverOLED #27: Линии без буфера

kostia-egik, 2023-03-17:
Режим без буфера.
Когда наклон линии ближе к вертикали (<45°), линия рассыпается на отдельные точки. Как я понял отрисовка следующей точки затирает предыдущую в этом столбце строки (у него отрисовка графики в текстовом режиме?).
У меня одного так?
oled 128х64, закупал пачку очень давно (лет 7 назад). SPI, 6 контактный.

GyverLibs, 2023-03-17:
Это особенность режима без буфера и внутреннего устройства дисплея

kostia-egik, 2023-03-17:
я для себя накидал костылей.
вроде бы работает.
// линия БЕЗ БУФЕРА
			void lineNoBuff(int x0, int y0, int x1, int y1, byte fill = 1) {
				_x = 0;
				_y = 0;
				if (x0 == x1) fastLineV(x0, y0, y1, fill);
				else if (y0 == y1) fastLineH(y0, x0, x1, fill);
				else {
					int sx, sy, e2, err;
					int dx = abs(x1 - x0);
					int dy = abs(y1 - y0);
					int preY = y0;
					sx = (x0 < x1) ? 1 : -1;
					sy = (y0 < y1) ? 1 : -1;
					err = dx - dy;
					if (dx > dy){
						for (;;) {
							dot(x0, y0, fill);
							if (x0 == x1 && y0 == y1) return;
							e2 = err<<1;
							if (e2 > -dy) { 
								err -= dy; 
								x0 += sx; 
							}
							if (e2 < dx) { 
								err += dx; 
								y0 += sy; 
							}
						}	
					}
					else{
						for (;;) {
							fastLineV(x0, y0, preY, fill);
							if (x0 == x1 && y0 == y1) {
								fastLineV(x0, preY, y1, fill);
								return;
							}
							e2 = err<<1;
							if (e2 > -dy) { 
								err -= dy; 
								x0 += sx;
								preY = y0+sy;
							}
							if (e2 < dx) { 
								err += dx; 
								y0 += sy; 
							}
							}
						}
					}
				}

GyverLibs, 2023-03-17:
Ну, это даже костылём страшно назвать) я потом когда нибудь планирую добавить буфер "последнего столбика" , с ним можно будет рисовать более свободно

kostia-egik, 2023-03-17:
чего могем.
Все равно эта конструкция работает только с одной линией. А если их много, без хотя бы локального буфера к сожалению не обойтись.

kostia-egik, 2023-03-17:
Ну, в конкретном случае есть огромный косяк от меня.
Не разобравшись в программной части, спаял все железки и запихал в корпус. Весьма замысловатый с плотной компоновкой и сложно разбираемый. Теперь вот выкручиваюсь, а знаний не хватает, будет мне урок.