GyverLibs/GyverOLED #30: SPI SCK and MOSI change

AndruKrug, 2023-04-24:
Добавьте, пожалуйста, возможность изменения адресов SCK и MOSI для SPI интерфейса.
У меня используются другие контакты, не смог разобраться как изменить.
"дисплей подключается D0 к SCK, D1 к MOSI"
П.С. не нашёл как буквы Ш и Щ отображаются?

GyverLibs, 2023-04-24:
В текущей версии дисплей работает только с аппаратным SPI

AndruKrug, 2023-04-24:
Спасибо за быстрый ответ.
В ESP32 есть 2 SPI интерфейса VSPI and HSPI, как их можно использовать?
Вроде они оба аппаратные.

GyverLibs, 2023-04-24:
```cpp
#define HSPI_MISO   12
#define HSPI_MOSI   13
#define HSPI_SCLK   14
#define HSPI_SS     15

// ...
SPI.begin(HSPI_SCLK, HSPI_MISO, HSPI_MOSI, HSPI_SS);
```

AndruKrug, 2023-04-24:
Спасибо, попробую