GyverLibs/GyverOLED #35: GyverOLEDMenu

gewisser, 2023-10-28:
Если кому нужно будет организовать меню, например, настроек - можете глянуть тут либу: https://github.com/gewisser/GyverOLEDMenu 

https://youtu.be/PtpRn7imspU

Она основана на GyverOLED. 

Если будет потребность, то просите гайвера слить её к себе :)

LemanRus, 2024-05-28:
Некорректно работает с OLED_NO_BUFFER, каждая следующая строка меню частично перекрывает предыдущую.

GyverLibs, 2024-05-28:
@LemanRus а чего ты мне по-английски то в соседнем ишью пишешь?))

LemanRus, 2024-05-30:
Был там в теме зарубежный гость, ну и мало ли, может та либа в мире популярна)

LemanRus, 2024-05-30:
> это корректное поведение для `OLED_NO_BUFFER`

Получается, для нормального использования придётся жертвовать нехило SRAM, на AVR уже придётся выкручиваться.

GyverLibs, 2024-05-30:
Ну, либо писать свою программу с осознанием особенностей дисплея и того, как работает no buffer. Также в библиотеке есть возможность выделить небольшой буфер динамически и выводить через него

LemanRus, 2024-05-30:
Метод update() с координатами, правильно понимаю?

Просто уже при создании объекта дисплея 128х64 с буфером он уже съедает килобайт памяти. Или его можно при выполнении программы назначить на часть дисплея?

Если можно, какой-нибудь небольшой код для наглядности)

GyverLibs, 2024-05-30:
Где то должен быть localbuffer или что то такое, в примерах должно быть

LemanRus, 2024-05-31:
```c++
  oled.clear();
  oled.createBuffer(64 - 20, 32 - 20, 64 + 20, 32 + 20);
  oled.circle(64, 32, 20, OLED_FILL);
  oled.sendBuffer();
  oled.update();
```

Выдернул кусочек из примера, оно? Будет по аналогии корректно работать с GyverOLED<SSD1306_128x64, OLED_NO_BUFFER>?