GyverLibs/GyverOLED #36: NO_BUFFER FILL

VladKolerts, 2023-11-29:
```
...
371#    void dot(int x, int y, byte fill = 1) { 
...
381# sendByte(1 << (y & 0b111));    // задвигаем 1 на высоту y
..

```
а почему собственно "1", а не "fill" ?
из-за  этого  в режиме без буффера нельзя флаг fill не работает

GyverLibs, 2024-03-26:
спасибо, поправил

v-ein, 2025-03-09:
Такая же проблема в `fastLineH` и, как следствие, в `line` для `y0 == y1`.

GyverLibs, 2025-03-21:
Проверено и работает?

v-ein, 2025-03-27:
Вот такой вариант проверен и работает:

```cpp
    void fastLineH(int y, int x0, int x1, byte fill = 1) {
        // ...
            byte data = !!fill << (y & 0b111);
```

Хотя можно как-то оптимальнее, наверное, - `(fill & 1)` или вообще `fill? (0b1 << (y & 0b111)) : 0`. Наверняка можно сэкономить пару байтиков :). Надо бы в генерируемый код заглянуть, но я так глубоко лезть не буду.