GyverLibs/GyverOLED #38: is % symbol supported?

raschiapedane, 2023-12-26:
I'm formatting a string containing a percentage symbol. Printing it using Serial.printf provides the following output

```
Temp: 18.42 C
R.Hum: 66.73 %
Press: 914.75 hPa
Brightness: 7
```

on the OLED display I see the same output without %

```
Temp: 18.42 C
R.Hum: 66.73
Press: 914.75 hPa
Brightness: 7
```

is the % symbol supported?

GyverLibs, 2023-12-26:
Yes, should be supported

closed 2024-01-25 by GyverLibs

Olejan, 2025-02-12:
To use % symbol use % prefix. For instance:
```
void draw_humidity(int x, int y, float humidity)
{
    char str[32];
    snprintf(str, sizeof(str), "R.Hum: %.2f %%", humidity);
    oled.setCursor(x, y);
    oled.print(str);
    oled.update();
}
```