GyverLibs/GyverOLED #40: Symbol degree ° are supported?

NickyX3, 2024-01-20:
Alex, i try out degree symbol ° (ALT+0176), but out "a" symbol
Алекс, я попробовал вывести символ градуса ° (ALT+0176), но выводится мелкая буква "а"

```
String text = bme.readTemperature() + "°C";
oled.setScale(2);
oled.setCursor(0,0);
oled.print(text);
```

Ожидаемое/Expected 27.74°C
Выводится/Output 27.74aC

GyverLibs, 2024-01-20:
Only ascii and Cyrillic is supported

closed 2024-01-25 by GyverLibs

Olejan, 2025-02-12:
> Ожидаемое/Expected 27.74°C Выводится/Output 27.74aC

You can replace any non usable symbol in the _charMap array onto degree code. For instance I do not use this symbol " ` "

```
  //{0x00, 0x00, 0x03, 0x04, 0x00}, // ` 0x60 96
  {0x00, 0x06, 0x09, 0x09, 0x06}, // degree 0x60 96
```

_charMap is located in the "src" directory.