GyverLibs/GyverOLED #46: Некорректная работа на платах ESP32-C3

Krypt0nC0R3, 2024-04-19:
Вместо вывода текста и графики на экране появляются точки в околопроизвольных местах. На библиотеке адафруита работает корректно, т.е. проблема не в дисплее.

Видео с демонстрацией проблемы: https://youtu.be/uwM4ZmfPJc0

GyverLibs, 2024-04-19:
i2c правильно настроен? Скорость какая?

Krypt0nC0R3, 2024-04-19:
i2c никак дополнительно не настраивал, все параметры по умолчанию. Скетч oled_demo с единственным изменением - высота экрана указана 32 вместо 64

Krypt0nC0R3, 2024-04-19:
Поглядел быстро библиотеку адафруита. Они при работе с I2C вообще не используют режим одной команды (OLED_ONE_COMMAND_MODE). Они всегда перед отправкой данных шлют 0x00. Может в этом дело?

Mebius84, 2026-08-29:
> Поглядел быстро библиотеку адафруита. Они при работе с I2C вообще не используют режим одной команды (OLED_ONE_COMMAND_MODE). Они всегда перед отправкой данных шлют 0x00. Может в этом дело?

GyverOLED<SSD1306_128x64, OLED_BUFFER> oled;
использовал такой вариант, отлично работает, китайская esp32c3 superMini