GyverLibs/GyverOLED #53: OLED i2c display reset issue

Luismiguelfco, 2024-09-19:
Hello Colleagues,

@AlexGyver Thank you for sharing this amazing library.

I'm using the 1.6.1 Gyveroled Arduino library version, but I'm facing problems with the Raspberry Pi Pico RP2040. 

I'm using Arduino IDE, Display I2C OLED 2,42" SSD1306 128x64, 4 pins (VCC,GND,SCL, and SDA).

The code (even the library examples, as oled_demo.ino) almost all the time works fine. But sometimes when I reset the Pico, the display doesn't wake up. If I turn off the Pico and display and turn it on again after 8 seconds, the oled display wakes up. Even if i press the PICO reset button, nothing happens.

I'm searching for answers in Google, and i found similar issues with i2C 4-pin OLED displays. 
It looks like one of those restarts faster than another, so it doesn't sync. In the link below, the guy solved the problem inserting a 25-ms delay in the start of code (I don't know exactly where).

[link](https://forum.arduino.cc/t/oled-reset-issue/89734?page=2)

I'm afraid because I made a big project with this hardware, and now I have this issue, and this is the only OLED library that works with this display and Pico.

Your support will be useful.

Thank you in advance.

GyverLibs, 2024-09-27:
Hi, try to add delay in beginning of setup

Luismiguelfco, 2024-09-28:
Hi!

It did not work.

I tryed with delay and delayMicroseconds of 25,100 and 500ms and i got the same issue.

Below you can see the code:
`

#include <SimpleTimer.h>
#include <GyverOLED.h>

GyverOLED<SSD1306_128x64, OLED_BUFFER> oled; // with buffer

#define Play_bt 1
#define Pause_bt 2
#define Reset_bt 3
#define Charge_pin 6
#define Bat_meter 26
#define Relay_pin 8
 

 
volatile int Segundos = 0;
volatile int Minutos = 0;
volatile int Horas = 0;
volatile int bat_percent = 0 ;
volatile bool play = 0;
volatile bool Paused = 1;
volatile bool Reseted = 0;
bool Charging = 0;
bool warning_flag = 0;
volatile float voltage = 0;

float v_min = 1.909;
float v_max = 2.273;
float perc_min = 0;
float perc_max = 100;

 
SimpleTimer timer;

 
    void setup() {
          delay (100);
          oled.init();  
          oled.clear();
          oled.update();
          

          pinMode (Bat_meter, INPUT);
          pinMode (Play_bt, INPUT);
          pinMode (Pause_bt, INPUT);
          pinMode (Reset_bt, INPUT);

          attachInterrupt(Pause_bt,Pause,FALLING);
          attachInterrupt(Reset_bt,Reset,FALLING);
          
          attachInterrupt(Play_bt, Play,RISING);
          
          timer.setInterval(1000, contaTempo);
          timer.setInterval(1200, get_percent);
          draw_timer ();
          }`

Thank you for your support

GyverLibs, 2024-09-28:
So where should be that delay?

Luismiguelfco, 2024-10-11:
> So where should be that delay?

I don't know :(

But I tried in a Arduino Nano, and the same happens. :(