GyverLibs/GyverOLED #56: Обрезанные строки, если выводить их со сдвигом по оси Y

Olejan, 2025-02-12:
Привет!
Есть задача вывести 6 строк на монохромный экран 128*64. С подключением вопросов нет, использую буфер. Вывожу 6 строк на экран. Если выводить их в каждую строку из восьми имеющихся, то всё рисуется прекрасно, но снизу остаётся пустое место из двух неиспользуемых строк. Задаю промежуток в 3 пикселя между строками, устанавливая позицию курсора функцией setCursorXY. Текст выводится как положено со смещением по оси Y, но часть второй и пятой строки обрезается. Как это побороть?

[image]

Вот тестовый код:
```
#include <GyverOLED.h>

GyverOLED<SSD1306_128x64, OLED_BUFFER> oled;

void setup() {
    oled.init();
    oled.clear();
}

void test_6_strings()
{
    const int symbol_height = 8, space_height = 3;
    oled.clear();
    for (int i = 0; i < 6; i++)
    {
        oled.setCursorXY(0, i * (symbol_height + space_height));
        oled.printf("Тестовая СТРОКА %d", i);
    }
    oled.update();
}

void loop() {
    static uint32_t ms = 0;
    if (millis() - ms > 1000)
    {
        ms = millis();
        test_6_strings();
    }
}
```

GyverLibs, 2025-02-12:
Возможно сменить режим текста на ADD

Olejan, 2025-02-12:
> Возможно сменить режим текста на ADD

Да, помогло `oled.textMode(0);`? благодарю!

closed 2025-02-15 by GyverLibs