GyverLibs/GyverOLED #59: Text is invisible if "out of the screeen" and Wasting the bottom line(s) of pixels

ShaggyDog18, 2025-02-27:
The embedded font is 6x8 pixels with the blank bottom line of pixels serving as a vertical text separator.
As a result, printing a text at the very bottom of the screen "wastes" the last line of pixels. If the text is moved one pixel down trying to use the last blank line of pixels, it becomes "out of the screen" and is not displayed , i.e. it does not appear at the screen. Not sure if this is a bug or a feature… 
That could be ignored for a large screen and a text with the scale of 1. Though, for a small narrow 124x32 display and a text with the scale of 2 that makes the bottom two lines of pixels unusable (which is 6% of the display!), i.e. the text cannot be shifted down to occupy these two last lines.
The potential solution: making the text printable at any place of the display (vertically) even if it is partially out of the screen.
Thank you!

GyverLibs, 2025-02-27:
Display in buffer mode or no buffer?

ShaggyDog18, 2025-02-27:
in buffer mode:
`GyverOLED<SSD1306_128x32, OLED_BUFFER> _display;`

ShaggyDog18, 2025-02-27:
here is simplified test:
```
#include <GyverOLED.h>
GyverOLED<SSD1306_128x32, OLED_BUFFER> _display;

void setup(void) {
  _display.init();
  _display.clear();
  _display.textMode( BUF_ADD );

  _display.setScale(2);

  _display.setCursorXY(0, 0);
  _display.print( "Line 1 TEST");

  _display.setCursorXY(0, 17);
  _display.print( "Line 2 TEST");   // line is not on the screen unless Y = 16
  

  _display.fastLineH( 30, 5, 120 ); 
  _display.fastLineH( 31, 0, 127 ); 

  _display.update();
}

void loop(void){}
```

ShaggyDog18, 2025-02-27:
textMode...  does not help with any value... 
The Line 2 from the above example is not visible, it is "out of screen" if  `setCursorXY(0, 17)`
The text is on the screen with `setCursorXY(0, 16)` but with 2 bottom "empty" lines that cannot be used in text mode....

ShaggyDog18, 2025-02-27:
Proposed workaround: comment line 285 in  GyverOLED.h :
`//if (_y + _scaleY > _maxY + 1) data = 0;  // дисплей переполнен`

ShaggyDog18, 2025-03-10:
Any thoughts about the proposed solution (above)?